stereolabs/zed-isaac-sim #6: Incompatible with Isaac Sim 4.0

mjshiggins, 2024-07-15:
Looks like Nvidia may have changed their directory structure requirements for Isaac Sim 4.0. The Extension window freezes on load. Will do a bit of troubleshooting myself.

[Warning] [omni.ext.plugin] [ext: .git] Extensions config 'extension.toml' doesn't exist '[REMOVED LOCAL REPO PATH]/zed-isaac-sim/.git' or '[REMOVED LOCAL REPO PATH]/zed-isaac-sim/.git/config'

mjshiggins, 2024-07-15:
For anyone else that gets stuck with a frozen sim window, you need to remove the ZED ext entry in ~/.local/share/ov/data/Kit/Isaac-Sim/4.0/user.config.json

mattrouss, 2024-08-29:
Hi @mjshiggins, support has been added with this PR #9.
Please try it out, further updates for Isaac Sim 4.0 will be available on the branch [release/4.0](https://github.com/stereolabs/zed-isaac-sim/tree/release/4.0)

mattrouss, 2024-11-20:
Closing as support for Isaac Sim 4.0 has been deployed.

closed 2024-11-20 by mattrouss