stereolabs/zed-isaac-sim #8: Incompatible with Isaac Sim 4.1

marmik18, 2024-08-08:
Error while loading the extension

```bash
[52.858s] [ext: sl.sensor.camera-1.1.0] startup
[sl.sensor.camera] SlSensorCameraExtension startup
2024-08-08 15:42:22 [52,859ms] [Error] [omni.graph] Error executing python callback sl.sensor.camera.ZED_Camera.initialize_type
```

Error while adding `action graph` node .
```bash
2024-08-08 15:53:18 [708,769ms] [Error] [omni.graph.core.plugin] Node::getAttribute() - attribute /World/ActionGraph/zed_camera_streamer.inputs:fps not found
2024-08-08 15:53:18 [708,769ms] [Error] [omni.graph] Error executing python callback sl.sensor.camera.ZED_Camera.initialize
2024-08-08 15:53:19 [709,584ms] [Error] [omni.ui.python] JSONDecodeError: Expecting value: line 1 column 1 (char 0)

At:
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/kit/python/lib/python3.10/json/decoder.py(355): raw_decode
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/kit/python/lib/python3.10/json/decoder.py(337): decode
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/kit/python/lib/python3.10/json/__init__.py(346): loads
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/extscache/omni.graph.ui-1.70.0+106.0.0.lx64.r.cp310/omni/graph/ui/_impl/omnigraph_attribute_base.py(238): _get_default_value
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/kit/kernel/py/carb/profiler/__init__.py(86): wrapper
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/extscache/omni.graph.ui-1.70.0+106.0.0.lx64.r.cp310/omni/graph/ui/_impl/omnigraph_attribute_base.py(668): _update_value_objects
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/extscache/omni.graph.ui-1.70.0+106.0.0.lx64.r.cp310/omni/graph/ui/_impl/omnigraph_attribute_base.py(623): _update_value
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/extscache/omni.graph.ui-1.70.0+106.0.0.lx64.r.cp310/omni/graph/ui/_impl/omnigraph_attribute_models.py(48): get_value_as_string
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/kit/exts/omni.kit.property.usd/omni/kit/property/usd/usd_property_widget_builder.py(1366): tftoken_builder
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/extscache/omni.graph.ui-1.70.0+106.0.0.lx64.r.cp310/omni/graph/ui/_impl/omnigraph_attribute_builder.py(516): _target_map_wrapper_builder
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/extscache/omni.graph.ui-1.70.0+106.0.0.lx64.r.cp310/omni/graph/ui/_impl/omnigraph_attribute_builder.py(248): build
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/extscache/omni.graph.ui-1.70.0+106.0.0.lx64.r.cp310/omni/graph/ui/_impl/compute_node_widget.py(417): build_property_item
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/kit/exts/omni.kit.property.usd/omni/kit/property/usd/usd_property_widget.py(385): _build_nested_group_props
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/kit/exts/omni.kit.property.usd/omni/kit/property/usd/usd_property_widget.py(450): _build_nested_group_frame
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/kit/exts/omni.kit.property.usd/omni/kit/property/usd/usd_property_widget.py(428): _build_nested_group_frame
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/kit/exts/omni.kit.property.usd/omni/kit/property/usd/usd_property_widget.py(476): build_nested_group_frames
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/kit/exts/omni.kit.property.usd/omni/kit/property/usd/usd_property_widget.py(532): build_items
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/kit/exts/omni.kit.window.property/omni/kit/window/property/templates/simple_property_widget.py(262): _build_frame
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/kit/exts/omni.kit.window.property/omni/kit/window/property/templates/simple_property_widget.py(281): _delayed_rebuild
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/kit/exts/omni.kit.window.property/omni/kit/window/property/templates/simple_property_widget.py(48): wrapper
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/kit/python/lib/python3.10/asyncio/events.py(80): _run
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/kit/python/lib/python3.10/asyncio/base_events.py(1909): _run_once
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/kit/extscore/omni.kit.async_engine/omni/kit/async_engine/async_engine.py(186): run_once
  /home/username/.local/share/ov/pkg/isaac-sim-4.1.0/kit/extscore/omni.kit.async_engine/omni/kit/async_engine/async_engine.py(251): <lambda>
 ```
 
Also, the node has missing inputs like fps, port, serial number, etc as shown in the image below.
[image]

strangeloveindustries, 2024-08-09:
Same issue here as well. I actually came to create this issue. Glad I'm not the only one...

mattrouss, 2024-11-20:
Hi @marmik18,

Version 2.0.0 of the isaac sim extension (https://github.com/stereolabs/zed-isaac-sim/tree/v2.0.0) has been tested and is compatible with Isaac Sim 4.1.
Please reopen the issue if you find more issues with Isaac Sim 4.1

closed 2024-11-20 by mattrouss