stereolabs/zed-isaac-sim #17: free(): invalid pointer when starting simulation including Zed camera streamer

TannerGilbert, 2025-02-12:
I'm currently working on a robotics project featuring the Zed 2i camera and wanted to simulate the robot inside Isaac Sim. When starting the simulation with the Zed camera streamer as described in the [documentation](https://docs.stereolabs.com/isaac-sim/setting_up_zed_isaac_sim/#build-the-action-graph) through the simulation crashes with the below error.

```
2025-02-11 19:09:20 [47,972ms] [Warning] [omni.hydra] Mesh '/World/ZED_X/base_link/ZED_X/ZED_X' has corrupted data in primvar 'normal': buffer size 35666 doesn't match expected size 153573 in facevarying primvars

2025-02-11 19:09:47 [75,360ms] [Warning] [omni.graph.tools._impl.deprecate] internal_state() is deprecated: 'internal_state' has been deprecated: use 'per_instance_state' or 'shared_state'
  File "/home/user/.local/share/ov/data/Kit/Isaac-Sim/4.2/exts/3/sl.sensor.camera-2.0.0+wx64.lx64.r.cp31/sl/sensor/camera/ogn/SlCameraStreamerDatabase.py", line 233, in compute
    return SlCameraStreamerDatabase.NODE_TYPE_CLASS.compute(db)
  File "/home/user/.local/share/ov/data/Kit/Isaac-Sim/4.2/exts/3/sl.sensor.camera-2.0.0+wx64.lx64.r.cp31/sl/sensor/camera/ogn/nodes/SlCameraStreamer.py", line 158, in compute
    if db.internal_state.initialized is False:

[Streaming] Use Transport layer mode : 32659
[Streaming] Use SHM-Boost IPC backend
free(): invalid pointer
2025-02-11 19:09:49 [0ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash] A crash has occurred.  If a debugger should be attached, please set the '/crashreporter/debuggerAttachTimeoutMs' setting to a timeout in milliseconds.  This can be used to allow the crash reporter to wait for up to that long for a debugger to attach before processing or sending the crash report.
2025-02-11 19:09:49 [41ms] [Warning] [carb.crashreporter-breakpad.plugin] [crash] compressing 3 files into '/home/user/.local/share/ov/data/Kit/Isaac-Sim/4.2/a9bce754-183b-41f9-1223c49c-bb36be33.dmp.zip' for upload.
2025-02-11 19:09:50 [455ms] [Info] [carb.crashreporter-breakpad.plugin] attempting to delete the file '/home/user/.local/share/ov/data/Kit/Isaac-Sim/4.2/a9bce754-183b-41f9-1223c49c-bb36be33.dmp' with 10 retries remaining.
2025-02-11 19:09:50 [466ms] [Info] [carb.crashreporter-breakpad.plugin] attempting to delete the file '/home/user/.local/share/ov/data/Kit/Isaac-Sim/4.2/a9bce754-183b-41f9-1223c49c-bb36be33.dmp.toml' with 10 retries remaining.
```

Running the real Zed 2i camera on the same system works without issues (ZED Diagnostics shows no errors). As the simulation crashes directly after starting I find this quite hard to debug. Any tips

System:
- Ubuntu 22
- CPU: Ryzen Threadripper
- GPU: NVIDIA RTX 6000

TannerGilbert, 2025-02-15:
Similarly to #15 downgrading solved the issue. In my case I downgraded to 4.1.4. I'll still keep this issue open as the issue still persist with the current version (4.2.5).

mattrouss, 2025-02-25:
A fix #18  was merged in the ZED Isaac Sim extension 2.0.1, please let me know if you are still able to reproduce

TannerGilbert, 2025-02-26:
Thanks addressing the problem. I'll try to test the new version this week and will get back to you if it fixes this issue.

TannerGilbert, 2025-02-26:
Just tested it. Above described error seems fixed with #18. @mattrouss thanks for the fix.

closed 2025-02-26 by TannerGilbert