stereolabs/zed-isaac-sim #21: Isaac sim 4.5, no zed camera streamer found

nuitvolgit, 2025-03-10:
Hello, 

I use Issac sim 4.5 and followed this tutorial.
https://www.stereolabs.com/docs/isaac-sim/setting_up_zed_isaac_sim

I was able to find zed camera extention from third party of extension manager.
However, when I enable the extention, I gout the following log:

2025-03-10 08:04:36 [1,209,605ms] [Warning] [omni.graph.tools._impl.internal.extension_contents_1_18] Exception while importing file /home/user/.local/share/ov/data/Kit/Isaac-Sim Full/4.5/exts/3/sl.sensor.camera-2.0.1+106.1.0.wx64.lx64.r.cp310/sl/sensor/camera/ogn/nodes/SlCameraStreamer.py: Traceback (most recent call last):
  File "/home/user/isaacsim/extscache/omni.graph.tools-1.79.1+d02c707b/omni/graph/tools/_impl/internal/extension_contents_1_18.py", line 366, in do_import_node_definition
    node_module = load_module_from_file(node_module_name, node_file)
  File "/home/user/isaacsim/extscache/omni.graph.tools-1.79.1+d02c707b/omni/graph/tools/_impl/internal/file_utils.py", line 238, in load_module_from_file
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/user/.local/share/ov/data/Kit/Isaac-Sim Full/4.5/exts/3/sl.sensor.camera-2.0.1+106.1.0.wx64.lx64.r.cp310/sl/sensor/camera/ogn/nodes/SlCameraStreamer.py", line 7, in <module>
    from omni.isaac.core_nodes.bindings import _omni_isaac_core_nodes
ModuleNotFoundError: No module named 'omni.isaac.core_nodes.bindings'

And I couldn't find the zed camera streamer node from the action graph search.

Is this version problem? If it is, is there any update schedule for this?

Thanks,

mattrouss, 2025-03-20:
Hi @nuitvolgit, we have just released the extension v3.0.0 which supports Isaac Sim 4.5, please let me know if you have any issue with the release.

Z3ZEL, 2025-03-21:
Hi,

Unfortunately, despite the release of 3.0.0, it's still not working on 4.0.5. I got this error in the console when activating the extension:

```
2025-03-21 14:54:07  [Warning] [omni.graph.tools._impl.internal.extension_contents_1_18] Exception while importing file /home/esrodr/.local/share/ov/data/Kit/Isaac-Sim Full/4.5/exts/3/sl.sensor.camera-3.0.0+106.5.wx64.lx64.r.cp310/sl/sensor/camera/ogn/nodes/SlCameraStreamer.py: Traceback (most recent call last):
2025-03-21 14:54:07  [Warning] [omni.graph.tools._impl.internal.extension_contents_1_18]   File "/home/esrodr/isaacsim/extscache/omni.graph.tools-1.79.1+d02c707b/omni/graph/tools/_impl/internal/extension_contents_1_18.py", line 366, in do_import_node_definition
2025-03-21 14:54:07  [Warning] [omni.graph.tools._impl.internal.extension_contents_1_18]     node_module = load_module_from_file(node_module_name, node_file)
2025-03-21 14:54:07  [Warning] [omni.graph.tools._impl.internal.extension_contents_1_18]   File "/home/esrodr/isaacsim/extscache/omni.graph.tools-1.79.1+d02c707b/omni/graph/tools/_impl/internal/file_utils.py", line 238, in load_module_from_file
2025-03-21 14:54:07  [Warning] [omni.graph.tools._impl.internal.extension_contents_1_18]     spec.loader.exec_module(module)
2025-03-21 14:54:07  [Warning] [omni.graph.tools._impl.internal.extension_contents_1_18]   File "<frozen importlib._bootstrap_external>", line 883, in exec_module
2025-03-21 14:54:07  [Warning] [omni.graph.tools._impl.internal.extension_contents_1_18]   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
2025-03-21 14:54:07  [Warning] [omni.graph.tools._impl.internal.extension_contents_1_18]   File "/home/esrodr/.local/share/ov/data/Kit/Isaac-Sim Full/4.5/exts/3/sl.sensor.camera-3.0.0+106.5.wx64.lx64.r.cp310/sl/sensor/camera/ogn/nodes/SlCameraStreamer.py", line 16, in <module>
2025-03-21 14:54:07  [Warning] [omni.graph.tools._impl.internal.extension_contents_1_18]     from sl.sensor.camera.pyzed_sim_streamer import ZEDSimStreamer, ZEDSimStreamerParams
2025-03-21 14:54:07  [Warning] [omni.graph.tools._impl.internal.extension_contents_1_18] ImportError: libsl_zed_sim_streamer.so: cannot open shared object file: No such file or directory
```
As you can see, 3.0.0 is indeed installed:
[image]
However, the extension is still not showing up:
[image]
I would be more than happy to help resolve this issue, as my project depends on it.

Cheers,

Esteban

Bvallon-sl, 2025-03-21:
Hi,

The name is "ZED camera streamer" without the "-", can you just look for "ZED".

You mentioned Isaac sim "4.0.5", is it a mistake? Did you mean 4.5? Because the issue was on Isaac Sim 4.5, not 4.0.5 as far as I know.

[image]

Best,

Z3ZEL, 2025-03-21:
Yes sorry, I meant 4.5.0

[image]

I've tried with 'ZED' but still nothing ... Are you on 4.5.0?

Bvallon-sl, 2025-03-24:
Yes I'm using Isaac Sim 4.5 on Windows.

Z3ZEL, 2025-03-24:
Ok, because according to the error message: `2025-03-24 09:24:28  [Warning] [omni.graph.tools._impl.internal.extension_contents_1_18] ImportError: libsl_zed_sim_streamer.so: cannot open shared object file: No such file or directory`, the `.so` file is an additional library for Linux. Maybe the `.dll` used for the Windows extension is working, but the Linux one is not?

Bvallon-sl, 2025-03-24:
Yes probably, thanks for sharing the error log, it's very helpful.
I'll test on a Ubuntu machine and come back to you.

mattrouss, 2025-03-24:
Hi @Z3ZEL,

I am not able to reproduct this on Ubuntu24 with the extension downloaded from Nvidia's registry.

Which version of Ubuntu are you using?
Also, have you imported the extension by cloning the repo or by installing it directly from the Extensions panel?

Z3ZEL, 2025-03-24:
Hi, thank you for taking the time to debug this issue.

Here is my system configuration:

## Report Details  
- **Generation Date:** 2025-03-24 11:37:04  

## Hardware Information:  
- **Hardware Model:** HP OMEN by HP 45L Gaming Desktop GT22-1xxx  
- **Memory:** 64.0 GiB  
- **Processor:** 13th Gen Intel® Core™ i9-13900K × 32  
- **Graphics Card:** NVIDIA GeForce RTX™ 4090  
- **Graphics Card 1:** NVIDIA GeForce RTX™ 4090  
- **Disk Capacity:** 2.0 TB  

## Software Information:  
- **Firmware Version:** F.11  
- **Operating System Name:** Ubuntu 24.04.2 LTS  
- **Operating System Build:** (null)  
- **Operating System Type:** 64-bit  
- **GNOME Version:** 46  
- **Windowing System:** X11  
- **Kernel Version:** Linux 6.11.0-19-generic

I'm using Isaac Sim 4.5 on a workstation installation (no Docker install), and I've installed the extension from the extension panel. I've also installed the ZED SDK on the computer.

Z3ZEL, 2025-03-26:
Hey, finally, I tried to do a clean install of everything, and it worked.  
Thank you for taking the time to help me.  
Cheers!

closed 2025-04-02 by mattrouss