stereolabs/zed-isaac-sim #23: Improvement of the Camera Releasing process

yizhouzhao, 2025-04-01:
Hi StereoLab, I am experiencing some GPU memory leak while using the ZedX camera streamer on Isaac Sim.

I found that the release function of the camera is too abrupt. It’s wrapped in a try-catch block without proper error handling:  [camera release](https://github.com/stereolabs/zed-isaac-sim/blob/9f2c853382429b2e08ca7ed0d444e2c8fa62962a/exts/sl.sensor.camera/sl/sensor/camera/nodes/SlCameraStreamer.py#L405)

After running the test several times, it seems the Zed SDK’s Python API may not be properly releasing the streaming process.

I am wondering if you could help verify how to release the camera streamer properly.

Bvallon-sl, 2025-04-02:
Hi,

I also noticed this behavior when working on the extension earlier this week. 
We are working on an important update with several improvements, including a fix for this leak.

Thanks for the report, and feel free to share your opinion on what we can improve and add to the extension in the future.

Best regards,

mattrouss, 2025-04-24:
Applied the fix in 1b3f0154f09608066e2bfaa42f1edf476e0c4957

@yizhouzhao could you take a look?

yizhouzhao, 2025-04-30:
Thanks a lot! I appreciate your help and I will close this issue.

closed 2025-04-30 by yizhouzhao