stereolabs/zed-isaac-sim #24: Extension compatibility with Isaac Sim 5.0.0

lesurJ, 2025-07-11:
Hello,

Thank you for the nice extension.
I am trying to integrate this extension in the new Isaac Sim 5.0.0 Developer Preview but without success. 
I am following [this tutorial](https://www.stereolabs.com/docs/isaac-sim/setting_up_zed_isaac_sim), but the extension cannot be installed. See error message below

```
2025-07-11T08:21:59Z [4,151,797ms] [Warning] [omni.kit.widget.cache_indicator.utils] No [paths] found in omniverse.toml
2025-07-11T08:24:55Z [4,327,996ms] [Warning] [omni.ext.plugin] [ext: exts] Extensions config 'extension.toml' doesn't exist '/home/user/GITS/simulatingsensors/zed-isaac-sim/exts' or '/home/user/GITS/simulatingsensors/zed-isaac-sim/exts/config'
2025-07-11T08:24:55Z [4,327,996ms] [Warning] [omni.ext.plugin] [ext: usd] Extensions config 'extension.toml' doesn't exist '/home/user/GITS/simulatingsensors/zed-isaac-sim/usd' or '/home/user/GITS/simulatingsensors/zed-isaac-sim/usd/config'
2025-07-11T08:25:45Z [4,377,865ms] [Error] [omni.ext.plugin] Can't pull extension: 'sl.sensor.camera-3.1.0', can't find it in the registry.
2025-07-11T08:26:52Z [4,445,198ms] [Error] [omni.ext.plugin] Can't pull extension: 'sl.sensor.camera-3.1.0', can't find it in the registry.

```

Is there a planned release date for porting this extension to the newest Isaac Sim version?  
Thanks

billqualls7, 2025-07-15:
the same

shaswat-shukla-cloud, 2025-10-28:
No idea about the warning. But, I think the first error can be resolved by creating a bin folder in the sl.camera.bridge folder and copying and pasting the libsl_zed file from the ZED SDK installation inside this bin folder. You can find all the ZED SDK files in the Program Files (x86) folder. There you can find the ZED SDK folder and inside there will be this file.